arduino-libraries/Arduino_USBHostMbed5 #16: Portenta Machine Control USB drive - Error mounting USB device -22

rayuwono, 2023-03-17:
Hello, I am trying to store and read files from a USB drive. I tested the example `DirList.ino` and uncomment the line ` mbed::DigitalOut otg(PB_14, 0);` as specified.

Here is my full code that I tested on Portenta Machine Control (PMC) with USB drive connected to it (just a regular USB stick, formatted to FAT32)

```cpp
/*
  Portenta - DirList

  The sketch shows how to mount an usb storage device and how to
  get a list of the existing folders and files.

  The circuit:
   - Portenta H7

  This example code is in the public domain.
*/

#include <Arduino_MachineControl.h>
#include <DigitalOut.h>
#include <FATFileSystem.h>
#include <Arduino_USBHostMbed5.h>

using namespace machinecontrol;

USBHostMSD msd;
mbed::FATFileSystem usb("usb");

// If you are using a Portenta Machine Control uncomment the following line
 mbed::DigitalOut otg(PB_14, 0);

void setup()
{
    Serial.begin(115200);
    while (!Serial)
        ;

    Serial.println("Starting USB Dir List example...");

    // if you are using a Max Carrier uncomment the following line
    // start_hub();

    while (!msd.connect()) {
        //while (!port.connected()) {
        delay(1000);
    }

    Serial.print("Mounting USB device... ");
    int err = usb.mount(&msd);
    if (err) {
        Serial.print("Error mounting USB device ");
        Serial.println(err);
        while (1);
    }
    Serial.println("done.");

    char buf[256];

    // Display the root directory
    Serial.print("Opening the root directory... ");
    DIR* d = opendir("/usb/");
    Serial.println(!d ? "Fail :(" : "Done");
    if (!d) {
        snprintf(buf, sizeof(buf), "error: %s (%d)\r\n", strerror(errno), -errno);
        Serial.print(buf);
    }
    Serial.println("done.");

    Serial.println("Root directory:");
    unsigned int count { 0 };
    while (true) {
        struct dirent* e = readdir(d);
        if (!e) {
            break;
        }
        count++;
        snprintf(buf, sizeof(buf), "    %s\r\n", e->d_name);
        Serial.print(buf);
    }
    Serial.print(count);
    Serial.println(" files found!");

    snprintf(buf, sizeof(buf), "Closing the root directory... ");
    Serial.print(buf);
    fflush(stdout);
    err = closedir(d);
    snprintf(buf, sizeof(buf), "%s\r\n", (err < 0 ? "Fail :(" : "OK"));
    Serial.print(buf);
    if (err < 0) {
        snprintf(buf, sizeof(buf), "error: %s (%d)\r\n", strerror(errno), -errno);
        Serial.print(buf);
    }
}

void loop()
{
}

```

However there is an error in the Serial Monitor. Any ideas?
```
Starting USB Dir List example...
Mounting USB device... Error mounting USB device -22
```

Zawi-Dzenklo, 2023-03-18:
> Hello, I am trying to store and read files from a USB drive. I tested the example `DirList.ino` and uncomment the line ` mbed::DigitalOut otg(PB_14, 0);` as specified.
> 
> Here is my full code that I tested on Portenta Machine Control (PMC) with USB drive connected to it (just a regular USB stick, formatted to FAT32)
> 
> ```c++
> /*
>   Portenta - DirList
> 
>   The sketch shows how to mount an usb storage device and how to
>   get a list of the existing folders and files.
> 
>   The circuit:
>    - Portenta H7
> 
>   This example code is in the public domain.
> */
> 
> #include <Arduino_MachineControl.h>
> #include <DigitalOut.h>
> #include <FATFileSystem.h>
> #include <Arduino_USBHostMbed5.h>
> 
> using namespace machinecontrol;
> 
> USBHostMSD msd;
> mbed::FATFileSystem usb("usb");
> 
> // If you are using a Portenta Machine Control uncomment the following line
>  mbed::DigitalOut otg(PB_14, 0);
> 
> void setup()
> {
>     Serial.begin(115200);
>     while (!Serial)
>         ;
> 
>     Serial.println("Starting USB Dir List example...");
> 
>     // if you are using a Max Carrier uncomment the following line
>     // start_hub();
> 
>     while (!msd.connect()) {
>         //while (!port.connected()) {
>         delay(1000);
>     }
> 
>     Serial.print("Mounting USB device... ");
>     int err = usb.mount(&msd);
>     if (err) {
>         Serial.print("Error mounting USB device ");
>         Serial.println(err);
>         while (1);
>     }
>     Serial.println("done.");
> 
>     char buf[256];
> 
>     // Display the root directory
>     Serial.print("Opening the root directory... ");
>     DIR* d = opendir("/usb/");
>     Serial.println(!d ? "Fail :(" : "Done");
>     if (!d) {
>         snprintf(buf, sizeof(buf), "error: %s (%d)\r\n", strerror(errno), -errno);
>         Serial.print(buf);
>     }
>     Serial.println("done.");
> 
>     Serial.println("Root directory:");
>     unsigned int count { 0 };
>     while (true) {
>         struct dirent* e = readdir(d);
>         if (!e) {
>             break;
>         }
>         count++;
>         snprintf(buf, sizeof(buf), "    %s\r\n", e->d_name);
>         Serial.print(buf);
>     }
>     Serial.print(count);
>     Serial.println(" files found!");
> 
>     snprintf(buf, sizeof(buf), "Closing the root directory... ");
>     Serial.print(buf);
>     fflush(stdout);
>     err = closedir(d);
>     snprintf(buf, sizeof(buf), "%s\r\n", (err < 0 ? "Fail :(" : "OK"));
>     Serial.print(buf);
>     if (err < 0) {
>         snprintf(buf, sizeof(buf), "error: %s (%d)\r\n", strerror(errno), -errno);
>         Serial.print(buf);
>     }
> }
> 
> void loop()
> {
> }
> ```
> 
> However there is an error in the Serial Monitor. Any ideas?
> 
> ```
> Starting USB Dir List example...
> Mounting USB device... Error mounting USB device -22
> ```
The Portenta Machine Control along with the Arduino giga has issues reading some types of USB sticks because there are fakes that are circulating around that have communication issues with these boards. I would first try formatting the flash drive as a smaller capacity(16mb should be a good start) with FAT file system instead of FAT32. If this still doesn't work,. then your best bet would probably be to buy a different brand USB flash drive.

aentinger, 2023-03-20:
Is your USB stick named `usb`? That's the most likely source of your mounting error. Alternately you can change the constant here, i.e.
```diff
-mbed::FATFileSystem usb("usb");
+mbed::FATFileSystem usb("myUsbStickName");
```

rayuwono, 2023-03-21:
> The Portenta Machine Control along with the Arduino giga has issues reading some types of USB sticks because there are fakes that are circulating around that have communication issues with these [...]

@Zawi-Dzenklo Following tests without code changes from above
#### Kingston 8GB DataTraveler G3

- USB 3.0
- FAT32
- name : `TEST (D:)`
- size: 7.56GB

- mounting error

```
Starting USB Dir List example...
Mounting USB device... Error mounting USB device -22
```

Repeat but now create the partition of 256MB

- USB 3.0
- FAT
- name : `TEST (D:)`
- size: 256MB
- mounting error

```
Starting USB Dir List example...
Mounting USB device... Error mounting USB device -22
```

#### Kingston 8GB 100 G2

- USB 2.0
- FAT32
- name : `TEST (D:)`
- size: 7.56GB
- No error!

```
Starting USB Dir List example...
Mounting USB device... done.
Opening the root directory... Done
done.
Root directory:
    System Volume Information
    .dropbox.device
2 files found!
Closing the root directory... OK
```

#### TwinMos USB2.0 Mobile Disk X4

- USB 2.0
- FAT
- name : `HIGHSCHOOL (D:)`
- size: 256MB
- No error!

```
Starting USB Dir List example...
Mounting USB device... done.
Opening the root directory... Done
done.
Root directory:
	...
    System Volume Information
    .dropbox.device
76 files found!
Closing the root directory... OK

```

> Is your USB stick named `usb`? That's the most likely source of your mounting error. Alternately you can change the constant here, i.e.
> 
> ```diff
> -mbed::FATFileSystem usb("usb");
> +mbed::FATFileSystem usb("myUsbStickName");
> ```

@aentinger 

Updated the name
```diff
-mbed::FATFileSystem usb("usb");
+mbed::FATFileSystem usb("TEST");

```

Tested
- USB 3.0
- FAT
- name : `TEST (D:)`
- size: 256MB
- mounting error
```
Starting USB Dir List example...
Mounting USB device... Error mounting USB device -22
```

alrvid, 2023-06-30:
The -22 error usually means that the FAT file system on the drive isn't correctly set up (already from the formatting, or from corruption somehow). Since you only get these errors with the Kingston 8GB DataTraveler G3, I would first suspect that particular thumb drive. I assume your drives were formatted on the same computer with the same operating system and settings. If not, it would be interesting to know the differences.

As for the name passed to mbed::FATFileSystem usb(), it doesn't have to match the name of the volume. It's used to signal where you want the drive mounted. If you pass "usb" it will be mounted at "/usb/" and a file in the root directory will be "/usb/file.txt". If you pass "mythumbdrive" the file will be at "/mythumbdrive/file.txt" instead, and so on.

Vasanth28897, 2023-12-19:
Hi, i want to read a battery data from DALY BMS thru USB-A port in the arduino portenta machine control act as a USBHost for this particular one. So, i noticed this `Arduino_USBHostMbed5` library could help on this, and i referred [this](https://github.com/maland16/daly-bms-uart/tree/main) to read the data from the Daly BMS., but first thing the arduino has to act as a usb host. Which example i can prefer on the given?