arduino-libraries/Arduino_USBHostMbed5 #21: Correct Version of Portenta MBED OS board files to use?

hhoppert, 2023-06-23:
It seems like the latest MBED releases removed USBHost support?  What version of the board files/OS should be used for this library?

hhoppert, 2023-06-24:
Alternatively, has anyone successfully used the samples on a Portenta Machine Control recently?

sebromero, 2023-10-19:
Hi @hhoppert
Have you been able to resolve this? This library should work with the Portenta Machine Control using the Arduino Portenta H7 core. We're using it in these two libraries that we recently released:

- https://github.com/arduino-libraries/Arduino_POSIXStorage/
- https://github.com/arduino-libraries/Arduino_UnifiedStorage

hhoppert, 2023-10-19:
I haven't tried recently, I will try again shortly.

closed 2025-07-01 by per1234