arduino-libraries/Arduino_USBHostMbed5 #30: GIGA R1 - Builds generate a lot of compiler warnings! (USBHALHost_STM.cpp)

KurtE, 2023-10-20:
This maybe is related to the crash as defined in #29

When you build a sketch such as the dirlist example for this library, it will generate a lot of compiler warnings.
Note: with the build caching, you may have to clear out the build cache to see these warnings after the first time you do a compile of the sketch.

```
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -Wall -Wextra -g3 -nostdlib "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_giga\\4.0.8\\variants\\GIGA/defines.txt" "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_giga\\4.0.8\\variants\\GIGA/cxxflags.txt" -MMD -mcpu=cortex-m7 -mfloat-abi=softfp -mfpu=fpv5-d16 -DARDUINO=10607 -DARDUINO_GIGA -DARDUINO_ARCH_MBED_GIGA -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_giga\\4.0.8\\cores\\arduino" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_giga\\4.0.8\\variants\\GIGA" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\Arduino_USBHostMbed5\\src" -DCM4_BINARY_START=0x60000000 -DCM4_BINARY_END=0x60040000 -DCM4_RAM_END=0x60080000 "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_giga\\4.0.8\\cores\\arduino/api/deprecated" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_giga\\4.0.8\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_giga\\4.0.8\\cores\\arduino" "@C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_giga\\4.0.8\\variants\\GIGA/includes.txt" "c:\\Users\\kurte\\Documents\\Arduino\\libraries\\Arduino_USBHostMbed5\\src\\targets\\TARGET_STM\\USBHALHost_STM.cpp" -o "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\0796A63DB76D221868BE353872FAE80B\\libraries\\Arduino_USBHostMbed5\\targets\\TARGET_STM\\USBHALHost_STM.cpp.o"
In file included from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\variants\GIGA/pinmode_arduino.h:23:0,
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/Arduino.h:26,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src/USBHost/USBHostConf.h:23,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.h:20,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.cpp:17:
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\variants\GIGA/mbed_config.h:260:0: warning: "MBED_CONF_PLATFORM_CALLBACK_NONTRIVIAL" redefined
 #define MBED_CONF_PLATFORM_CALLBACK_NONTRIVIAL                                  1                                                                                                // set by application[*]
 
In file included from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.h:20:0,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.cpp:17:
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src/USBHost/USBHostConf.h:21:0: note: this is the location of the previous definition
 #define MBED_CONF_PLATFORM_CALLBACK_NONTRIVIAL  0
 
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.cpp: In member function 'bool USBHostHub::connect(USBDeviceConnected*)':
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.cpp:125:45: warning: 'void rtos::ThisThread::sleep_for(uint32_t)' is deprecated: Pass a chrono duration, not an integer millisecond count. For example use `5s` rather than `5000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
         rtos::ThisThread::sleep_for(buf[5]*2);
                                             ^
In file included from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/rtos/include/rtos/rtos.h:30:0,
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/mbed.h:24,
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed.h:8,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src/USBHost/USBHostTypes.h:20,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.h:24,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.cpp:17:
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/rtos/include/rtos/ThisThread.h:216:6: note: declared here
 void sleep_for(uint32_t millisec);
      ^~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.cpp: In member function 'void USBHostHub::portReset(uint8_t)':
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.cpp:252:39: warning: 'void rtos::ThisThread::sleep_for(uint32_t)' is deprecated: Pass a chrono duration, not an integer millisecond count. For example use `5s` rather than `5000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
         rtos::ThisThread::sleep_for(10);
                                       ^
In file included from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/rtos/include/rtos/rtos.h:30:0,
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/mbed.h:24,
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed.h:8,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src/USBHost/USBHostTypes.h:20,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.h:24,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHostHub\USBHostHub.cpp:17:
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/rtos/include/rtos/ThisThread.h:216:6: note: declared here
 void sleep_for(uint32_t millisec);
      ^~~~~~~~~
In file included from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\variants\GIGA/pinmode_arduino.h:23:0,
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/Arduino.h:26,
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed.h:4,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src/USBHost/USBHostTypes.h:20,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src/USBHost/USBEndpoint.h:23,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:22:
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\variants\GIGA/mbed_config.h:260:0: warning: "MBED_CONF_PLATFORM_CALLBACK_NONTRIVIAL" redefined
 #define MBED_CONF_PLATFORM_CALLBACK_NONTRIVIAL                                  1                                                                                                // set by application[*]
 
In file included from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:22:0:
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src/USBHost/USBEndpoint.h:21:0: note: this is the location of the previous definition
 #define MBED_CONF_PLATFORM_CALLBACK_NONTRIVIAL  0
 
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp: In member function 'void USBEndpoint::init(HCED*, ENDPOINT_TYPE, ENDPOINT_DIRECTION, uint32_t, uint8_t, HCTD**)':
[... cut ...]  conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
                                         UNUSED(tmpreg); \
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_def.h:70:27: note: in definition of macro 'UNUSED'
 #define UNUSED(x) ((void)(x))
                           ^
```

(Truncated to fit the max size of post)

KurtE, 2023-10-20:
(continued from the first)
```
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.h:109:30: note: in expansion of macro '__HAL_RCC_GPIOA_CLK_ENABLE'
 #define USB_POWERPIN_CONFIG {__HAL_RCC_GPIOA_CLK_ENABLE();gpio_init_out_ex(&gpio_powerpin, PA_15, USB_POWER_OFF);}
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.h:342:5: note: in expansion of macro 'USB_POWERPIN_CONFIG'
     USB_POWERPIN_CONFIG;
     ^
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_rcc.h:2423:48: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
                                         UNUSED(tmpreg); \
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_def.h:70:27: note: in definition of macro 'UNUSED'
 #define UNUSED(x) ((void)(x))
                           ^
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.h:345:5: note: in expansion of macro '__HAL_RCC_SYSCFG_CLK_ENABLE'
     __HAL_RCC_SYSCFG_CLK_ENABLE();
     ^
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_rcc.h:1047:48: warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}'
                                         UNUSED(tmpreg); \
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_def.h:70:27: note: in definition of macro 'UNUSED'
 #define UNUSED(x) ((void)(x))
                           ^
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/Legacy/stm32_hal_legacy.h:3145:56: note: in expansion of macro '__HAL_RCC_USB1_OTG_HS_CLK_ENABLE'
 #define __HAL_RCC_USB_OTG_HS_CLK_ENABLE()              __HAL_RCC_USB1_OTG_HS_CLK_ENABLE()
                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.h:353:5: note: in expansion of macro '__HAL_RCC_USB_OTG_HS_CLK_ENABLE'
     __HAL_RCC_USB_OTG_HS_CLK_ENABLE();
     ^
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp: In function 'void HAL_HCD_EnableInt(HCD_HandleTypeDef*, uint8_t)':
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp:63:28: warning: unused variable 'USBx' [-Wunused-variable]
     USB_OTG_GlobalTypeDef *USBx = hhcd->Instance;
                            ^~~~
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp: In function 'void HAL_HCD_DisableInt(HCD_HandleTypeDef*, uint8_t)':
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp:70:28: warning: unused variable 'USBx' [-Wunused-variable]
     USB_OTG_GlobalTypeDef *USBx = hhcd->Instance;
                            ^~~~
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp: In function 'void HAL_HCD_HC_NotifyURBChange_Callback(HCD_HandleTypeDef*, uint8_t, HCD_URBStateTypeDef)':
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp:109:20: warning: enumeration value 'URB_IDLE' not handled in switch [-Wswitch]
             switch (urb_state) {
                    ^
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp:109:20: warning: enumeration value 'URB_NYET' not handled in switch [-Wswitch]
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp:109:20: warning: enumeration value 'URB_ERROR' not handled in switch [-Wswitch]
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp:109:20: warning: enumeration value 'URB_STALL' not handled in switch [-Wswitch]
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp: In member function 'void USBHALHost::resetRootHub()':
c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBHALHost_STM.cpp:337:36: warning: 'void rtos::ThisThread::sleep_for(uint32_t)' is deprecated: Pass a chrono duration, not an integer millisecond count. For example use `5s` rather than `5000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
     rtos::ThisThread::sleep_for(200);
                                    ^
In file included from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/rtos/include/rtos/rtos.h:30:0,
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/mbed.h:24,
[... cut ...]
                 from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed.h:8,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.h:20,
                 from c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:18:
C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.8\cores\arduino/mbed/rtos/include/rtos/Queue.h:311:13: note: declared here
     osEvent get(uint32_t millisec = osWaitForever)
             ^~~
Compiling core...

```

alrvid, 2023-10-27:
I just want you to know that we're aware of this. Some of the warnings are easy to fix, and others a bit more complicated. I'm keeping them in mind when trying to solve the crash bug in case they matter for that, but I'm also keeping them at a lower priority than the bug right now. Thanks for pointing it out though. I prefer if there are no warnings at all, even when all warnings are enabled, so the long term goal is to get rid of them all if possible.