arduino-libraries/Arduino_USBHostMbed5 #32: GIGA R1:  FTDI adapters do not return the correct Vendor and Product IDs

KurtE, 2023-10-30:
As I have mentioned on the forum:
https://forum.arduino.cc/t/can-not-find-examples-for-usbhostserial/1182486/17
I am trying to hack up some code similar to the USBHostSerial code that works with other USB to Serial adapters other than those which are CDC ACM devices.  Ones like FTDI, Prolific...

On the forum thread: I posted an example sketch which reproduces this, but just about any sketch that will print out the 
wrong IDS.  VID: 0103 PID: 0000

Where other code bases on some other Arduinos, plus Windows and Ubuntu return the IDs:
vid=403 pid=6001

Note: I just tried plugging it in to the GIGA board, with my Logic Analyzer hooked up to it, with my modified USB Analyzer and HLA, which helps reduce the data down...

The Packets that I received on the GIGA:
```
16.308830162 ; SETUP ; 0x0 ; 0x0 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x8] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x8 0x0
16.308907428 ; IN ; 0x0 ; 0x0 ;  ;  0x12 0x1 0x0 0x2 0x0 0x0 0x0 0x8
16.308963208 ; SETUP ; 0x0 ; 0x0 ; SET_ADDRESS I:0x0 L:0x0] ;  0x0 0x5 0x1 0x0 0x0 0x0 0x0 0x0
16.317414284 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x8] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x8 0x0
16.317496376 ; IN ; 0x0 ; 0x1 ;  ;  0x12 0x1 0x0 0x2 0x0 0x0 0x0 0x8
20.478046968 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x12] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x12 0x0
20.478125602 ; IN ; 0x0 ; 0x1 ;  ;  0x12 0x1 0x0 0x2 0x0 0x0 0x0 0x8
20.478314948 ; IN ; 0x0 ; 0x1 ;  ;  0x3 0x4 0x1 0x60 0x0 0x6 0x1 0x2
20.478336022 ; IN ; 0x0 ; 0x1 ;  ;  0x3 0x1
20.502892408 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - CONFIG #:0 I:0x0 L:0x9] ;  0x80 0x6 0x0 0x2 0x0 0x0 0x9 0x0
20.503027744 ; IN ; 0x0 ; 0x1 ;  ;  0x9 0x2 0x20 0x0 0x1 0x1 0x0 0xa0
20.50305689 ; IN ; 0x0 ; 0x1 ;  ;  0x2d
20.510192198 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - CONFIG #:0 I:0x0 L:0x20] ;  0x80 0x6 0x0 0x2 0x0 0x0 0x20 0x0
20.510326864 ; IN ; 0x0 ; 0x1 ;  ;  0x9 0x2 0x20 0x0 0x1 0x1 0x0 0xa0
20.510358858 ; IN ; 0x0 ; 0x1 ;  ;  0x2d 0x9 0x4 0x0 0x0 0x2 0xff 0xff
20.510387752 ; IN ; 0x0 ; 0x1 ;  ;  0xff 0x2 0x7 0x5 0x81 0x2 0x40 0x0
20.510416558 ; IN ; 0x0 ; 0x1 ;  ;  0x0 0x7 0x5 0x2 0x2 0x40 0x0 0x0

```
I don't have time today, but it almost looks like the device is still only returning a max of something like 8 bytes at a time...
And You are probably indexing off the end of the data, and maybe not waiting for the additional data packets to be returned.

If I remember correctly on the Teensy USBHost code, we first asked for the device descriptor, for its first 8 bytes, and then update the control endpoint to the size specified in those first 8 bytes and then asked for the whole endpoint...

AndrewCapon, 2023-10-30:
Are you using the most recent version from GitHub?

I had this 8 byte issue before, pretty sure using the version from GitHub fixed this.

AndrewCapon, 2023-10-30:
Pretty sure the last bis was not there:

```
                            devices[i].setSizeControlEndpoint(8);

                            devices[i].activeAddress(false);

                            // get first 8 bit of device descriptor
                            // and check if we deal with a hub
                            USB_DBG("usb_thread read device descriptor on dev: %p\r\n", &devices[i]);
                            res = getDeviceDescriptor(&devices[i], buf, 8);

                            if (res != USB_TYPE_OK) {
                                USB_ERR("usb_thread could not read dev descr");
                                continue;
                            }

                            // set size of control endpoint
                            devices[i].setSizeControlEndpoint(buf[7]);
```
Without that last `setSizeControlEndpoint` the size sticks at 8 when the enumerate core is run.

AndrewCapon, 2023-10-30:
Actually I think I'm wrong, I have been testing loads of midi devices and one of them returns the size requested in buf[7], so 8.

using instead:
```
devices[i].setSizeControlEndpoint(64);
```

And this device then enumerates.

AndrewCapon, 2023-10-30:
Incorrectly it seems, ignore me!

AndrewCapon, 2023-10-30:
Right, I think this is related to : https://github.com/arduino-libraries/Arduino_USBHostMbed5/issues/31

The device I have here wants to deal with 8 byte packs, I guess the same as your one.

So when it tries to get the full 18 bytes to get the VID and PID, this going to require 3 packets.

The existing code should request these three packets but it doesn't work for me here:

```
        if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) {
            switch (urb_state) {
                case URB_DONE:
#if defined(MAX_NOTREADY_RETRY)
                    td->retry = 0;
#endif
                    if (td->size >  max_size) {
                        /*  enqueue  another request */
                        td->currBufPtr += max_size;
                        td->size -= max_size;
                        length = td->size <= max_size ? td->size : max_size;
                        HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
                        HAL_HCD_EnableInt(hhcd, chnum);
                        return;
                    }
                    break;
                case  URB_NOTREADY:
#if defined(MAX_NOTREADY_RETRY)
                    if (td->retry < MAX_NOTREADY_RETRY) {
                        td->retry++;
#endif
                        // Submit the same request again, because the device wasn't ready to accept the last one
                        length = td->size <= max_size ? td->size : max_size;
                        HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
                        HAL_HCD_EnableInt(hhcd, chnum);
                        return;

                        return;
#if defined(MAX_NOTREADY_RETRY)
                    } else {
                        // MAX_NOTREADY_RETRY reached, so stop trying to resend and instead wait for a timeout at a higher layer
                    }
#endif
                    break;
            }
        }
```

AndrewCapon, 2023-10-30:
I just tried a prolific one I have here but it works, it is using 64 byte packets though.

KurtE, 2023-10-30:
I am running version 0.2.0 - and I don't see any other changes since then.  Mine is slightly mucked up for debug, plus maybe forced to not simple callbacks...

Looking at the data from Logic Analyzer:
```
20.478046968 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x12] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x12 0x0
20.478125602 ; IN ; 0x0 ; 0x1 ;  ;  0x12 0x1 0x0 0x2 0x0 0x0 0x0 0x8
20.478314948 ; IN ; 0x0 ; 0x1 ;  ;  0x3 0x4 0x1 0x60 0x0 0x6 0x1 0x2
20.478336022 ; IN ; 0x0 ; 0x1 ;  ;  0x3 0x1
```

The line:
` devices[i].setSizeControlEndpoint(buf[7]);
`
Set the size to 8  Valid sizes are ( 8, 16, 32, 64).  Wonder if your other devices have larger values...
When it asked for 0x12 (18) bytes it did properly return 3 packets.
Combining them we have:
```
 0        1      2    3    4    5    6    7     8     9    10   11    12   13  14   15  16   17 
0x12 0x1 0x0 0x2 0x0 0x0 0x0 0x8 0x3 0x4 0x1 0x60 0x0 0x6 0x1 0x2 0x3 0x1
```
So ID vendor(bytes 8-9): 0x403 and Product: 0x601

But it returned vid 0x103... Maybe like it skipped the middle packet.  Maybe assuming end point size of 16?

AndrewCapon, 2023-10-30:
Is the device you are having trouble with returning 8 in buf[7]?

What I was seeing with midi devices was that I was never getting > 64 bytes returned, so just a single packet. I was not seeing what you are seeing there with multiple packets with the existing code.

KurtE, 2023-10-30:
Yes it is returning 0x8 in buf[7]

Where as the PL2303 device the USB data looks like:
```
14.689485844 ; SETUP ; 0x0 ; 0x0 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x8] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x8 0x0
14.689516876 ; IN ; 0x0 ; 0x0 ;  ;  0x12 0x1 0x10 0x1 0x0 0x0 0x0 0x40
14.689572596 ; SETUP ; 0x0 ; 0x0 ; SET_ADDRESS I:0x0 L:0x0] ;  0x0 0x5 0x1 0x0 0x0 0x0 0x0 0x0
14.691096216 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x8] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x8 0x0
14.691125776 ; IN ; 0x0 ; 0x1 ;  ;  0x12 0x1 0x10 0x1 0x0 0x0 0x0 0x40
17.252858628 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x12] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x12 0x0
17.252888864 ; IN ; 0x0 ; 0x1 ; SET_PROTOCOL(0x0xb) ;  0x12 0x1 0x10 0x1 0x0 0x0 0x0 0x40 0x7b 0x6 0x3 0x23 0x0 0x4 0x1 0x2 0x0 0x1
17.253834594 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - CONFIG #:0 I:0x0 L:0x9] ;  0x80 0x6 0x0 0x2 0x0 0x0 0x9 0x0
17.253865 ; IN ; 0x0 ; 0x1 ; 0x32(0x0x2) ;  0x9 0x2 0x27 0x0 0x1 0x1 0x0 0x80 0x32
17.254269648 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - CONFIG #:0 I:0x0 L:0x27] ;  0x80 0x6 0x0 0x2 0x0 0x0 0x27 0x0
17.254299062 ; IN ; 0x0 ; 0x1 ; 0x32(0x0x2) ;  0x9 0x2 0x27 0x0 0x1 0x1 0x0 0x80 0x32 0x9 0x4 0x0 0x0 0x3 0xff 0x0 0x0 0x0 0x7 0x5 0x81 0x3 0xa 0x0 0x1 0x7 0x5 0x2 0x2 0x40 0x0 0x0 0x7 0x5 0x83 0x2 0x40 0x0 0x0
17.258187612 ; SETUP ; 0x0 ; 0x1 ; [SET_CONFIGURATION I:0x0 L:0x0] ;  0x0 0x9 0x1 0x0 0x0 0x0 0x0 0x0
```
So it has 64 byte control packets. And retrieved the proper VID/PID...

Before it crashes... Currently in the Red blink... 
But that is a different issue...

AndrewCapon, 2023-10-30:
So with 64 byte packets here with the existing code the issue I was seeing was when the configuration data was > 64 bytes, only 64 bytes were ever returned.

Changing to using HAL_HCD_HC_SubmitRequest() to request the full amount of data, say 101 bytes then fixed this issue for me. Which is what I was going on about in the other issue.

I now have a midi device that only uses 8 byte packets, this is not working correctly at all HAL_HCD_HC_SubmitRequest() is not returning all the data with an 8 byte packet. Interestingly if I set the packet size to 64 and then request 8 times more data I get the 64 byte packets with the correct 8 bytes at the beginning of each.

At the moment for me here I'm wondering about an underlying HAL issue with 8 byte packets.

KurtE, 2023-10-30:
Could be HAL or their library, not sure where the code is that detects that requested data exceeded the size of a packet, and then requests or waits for the right number of data packets to come back in this case and pack the data into the correct buffer... 

Let alone what is there any support anywhere in this library and/or HAL when I plug in a high speed device who typically might send 512 byte packets, although I think the control endpoint stays at max 64 byte...

AndrewCapon, 2023-10-30:
The code as it is is using the code I posted above to split the packets, this code is in this repository. So they never call HAL_HCD_HC_SubmitRequest() with greater than the packet size, they do all the work here.

This is the code I was pretty sure is not working, so I changed it to let HAL_HCD_HC_SubmitRequest() split the packets as it already does this for you and this solved my issues with 64 byte packets and everything now works. You can see what I did in the PR in the other issue: https://github.com/arduino-libraries/Arduino_USBHostMbed5/pull/33

Unfortunately HAL_HCD_HC_SubmitRequest(0 doesn't seem to be working with 8 byte packets, the top level code looks fine it is requesting the correct number of packets, something is going wrong though.

I hate USB stuff!

KurtE, 2023-11-01:
@alrvid - Not sure if you are the main person on this library?  

@AndrewCapon - I am guessing that the code is losing packets.

Note: I renamed the issue as I believe it is not specific to FTDI.  I am also not sure how specific this is to reading the device descriptor, or if it might manifest issues, whenever a response to a request might span multiple USB packets.  Or potentially if the device might generate a burst of data.

More specifics about this manifestation: 
When I plug in an FTDI adapter, The code does not properly return the 18 bytes of the device descriptor:
I know that the device is sending the 18 bytes, in three input packets:

The USB analyzer code I have running on my Saleae Logic Analyzer shows that: 
```
1.979299366 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x12] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x12 0x0
1.9867984699999999 ; IN ; 0x0 ; 0x1 ;  ;  0x12 0x1 0x0 0x2 0x0 0x0 0x0 0x8
1.9869933480000002 ; IN ; 0x0 ; 0x1 ;  ;  0x3 0x4 0x1 0x60 0x0 0x6 0x1 0x2
1.987014406 ; IN ; 0x0 ; 0x1 ;  ;  0x3 0x1
```
[image]

Probably does not show it very well, but there is about 8ms from the time of the request to the response.  You can see this by the areas of blue at the top which is from the HLA analyzer showing the packet data.  Zooming in on the 3 packet area:
[image]
You can see they are clumped pretty close together, especially the 2nd two.

Looking at debug output 
```
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1216]SETUP PACKET: 
80 6 0 1 0 0 12 0 
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1244]CONTROL setup stage USB_TYPE_IDLE
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1273]CONTROL READ stage USB_TYPE_IDLE
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1281]CONTROL READ SUCCESS [16 bytes transferred]
12 01 00 02 00 00 00 08 03 01 00 00 00 00 00 00 

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1314]CONTROL ack stage USB_TYPE_IDLE
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:957]DEVICE DESCRIPTOR(18):

12 01 00 02 00 00 00 08 03 01 00 00 00 00 00 00 00 00 

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:969]CLASS: 00 	 VID: 0103 	 PID: 0000
[USB_INFO: c:\Users\kurte\Documents\Arduino\libraries\GIGA_USBHostMBed5_devices\src\USBHostSerialDevice.cpp:165]VID: 103, PID: 0
```
You can see it completely missed the data of the 2nd IN packet.  And in fact used the first two bytes of the 3rd packet and assumed it received 8 bytes.  i.e. it did not look at the actual size of data in the IN packet. 

Interesting that the USBHost::getDeviceDescriptor call has parameter to return the size of data returned, the return value actually just returns the size of data requested (or size of a device descriptor).... 
So it prints out now 18 bytes....

As a test, I tried hacking the 2nd request to get the descriptor from asking for 18 bytes and instead only ask for 16.  And it returned all 0s after the first 8 bytes... Again, it lost the 2nd packet.

Guessing: that maybe you don't have a request queued up in time to receive the 2nd part of the data. 

Sorry, I don't know enough about this board, to understand it's lower level mechanisms on queueing up requests.  On some other systems, I believe there are ways to queue up multiple requests, such that when the hardware fills one, it is setup to handle the next, not sure if that is possible here or not.

KurtE, 2023-11-01:
It also appears to be failing with CH340G like devices such as:
https://www.sparkfun.com/products/15096

```
5.923490414 ; SETUP ; 0x0 ; 0x1 ; [GET_DESCRIPTOR - DEVICE #:0 I:0x0 L:0x12] ;  0x80 0x6 0x0 0x1 0x0 0x0 0x12 0x0
5.923526774 ; IN ; 0x0 ; 0x1 ;  ;  0x12 0x1 0x10 0x1 0xff 0x0 0x0 0x8
5.923555124 ; IN ; 0x0 ; 0x1 ;  ;  0x86 0x1a 0x23 0x75 0x54 0x2 0x0 0x2
5.92357572 ; IN ; 0x0 ; 0x1 ;  ;  0x0 0x1
```

Comes back
`12 01 10 01 FF 00 00 08 00 01 00 00 00 00 00 00 00 00 `
Instead of what it should be:
`12 01 10 01 ff 00 00 08 86 1a 23 75 54 02 00 02 00 01`

AndrewCapon, 2023-11-02:
Hi @KurtE ,

From my fumblings I have noticed lengths reported and number of bytes logged in the USB debug messages cannot be trusted!

I am sadly lacking a USB analyser, I do have a Saleae clone though, it is only 400mhz though. What is yours?

I am looking into the USB_NOTREADY/NAK thing, I am guessing the issue may be originating here. What I don't understand though is that devices should not NAK on Setup messages, so we should never see NAKs when getting the device description (but I am here). Is this your understanding as well?

KurtE, 2023-11-02:
Morning @AndrewCapon,

I also don't have a full USB analyzer.  Like a Beagle 12 (about $500) or 480 (about $1300 and up), which would be great for analyzing this stuff.  

I do however have a few Saleae logic analyzers.  The main one I use is their Pro 8. Which has gone up in price a lot since I purchased it...  You can still get the non-pro 8 as a student or as a hobbyist for about $250.

They come with a USB analyzer setup for LS and FS (but can not handle HS).  They have not yet updated it to support their newer High Level analyzer stuff, so I took a stab at it, which is working reasonably well for me.,,,

Note: I was hoping that your stuff in your PR  #33 would help with the stuff I am seeing.  

So far still no luck.  Will double check I put in the change correctly. So far still not getting the configuration descriptor for some of these boards... Will double check I have it in the same way you do.

AndrewCapon, 2023-11-02:
Afternoon @KurtE

Have you got a little breakout board for the usb signals then?

PR 33 would help I'm guessing with 64 byte packets because that works here, it doesn't work with 8 byte packets here though, I'm trying to get to the bottom of it. Unfortunately I have messed my back up so can't sit at the desk so am trying to use an old laptop with Remote Desktop so the going is slow :)

Basically what I am seeing is loads of NAKs, I can re-add in the code to handle NAKs which then means the descriptors work but it then goes wonky elsewhere.

KurtE, 2023-11-02:
Sorry about your back... 
[image]
Also my setup is sort of a mess, but on this board, I simply use a couple of the test clips (that came with the LAs) https://usd.saleae.com/collections/accessories/products/test-clips-93

The center two pins are the D+ and D-...

On my own boards, like the one in the picture (Micromod Teensy), I added breakout pins which are the two on the left of the USB Host connector, for those signals to make it easy.

I will try a bit more with your changes to see what is happening. 

This code looks considerably different than the code USBHostGiga,  So I might try a few more things out on that library to see how it differs on handling the packets.

I don't understand why the released code, Tries to restrict the request down to one packet and then try to start up new request for next packet...  I could be completely wrong, but I believe that the Channels on these boards are setup to handle transfers > packet_size... That is the OTG_HCTSIZx registers, look like they have fields for Transfer size, and then the 19 bits, then a packet count 10 bits.   

So I would think it should just handle it.

I believe that it funnels down to HAL_StatusTypeDef USB_HC_StartXfer(USB_OTG_GlobalTypeDef *USBx, USB_OTG_HCTypeDef *hc, uint8_t dma)

which does:
```
  if (hc->xfer_len > 0U)
  {
    num_packets = (uint16_t)((hc->xfer_len + hc->max_packet - 1U) / hc->max_packet);

    if (num_packets > max_hc_pkt_count)
    {
      num_packets = max_hc_pkt_count;
      hc->XferSize = (uint32_t)num_packets * hc->max_packet;
    }
  }
```

So wondering if I am reading the stuff wrong, or there were problems with it or ???

AndrewCapon, 2023-11-02:
The STM HAL stuff does handle multiple packets, the PR 33 I put in changes the "MBED" code to use this functionality.

The existing code does not though, the MBED code is never requesting more than one packet at a time, they do the handling of multiple packets.

If you look at the code I removed:
```
       if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) {
            switch (urb_state) {
                case URB_DONE:
#if defined(MAX_NOTREADY_RETRY)
                    td->retry = 0;
#endif
                    if (td->size >  max_size) {
                        /*  enqueue  another request */
                        td->currBufPtr += max_size;
                        td->size -= max_size;
                        length = td->size <= max_size ? td->size : max_size;
                        HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
                        HAL_HCD_EnableInt(hhcd, chnum);
                        return;
                    }
                    break;
                case  URB_NOTREADY:
#if defined(MAX_NOTREADY_RETRY)
                    if (td->retry < MAX_NOTREADY_RETRY) {
                        td->retry++;
#endif
                        // Submit the same request again, because the device wasn't ready to accept the last one
                        length = td->size <= max_size ? td->size : max_size;
                        HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
                        HAL_HCD_EnableInt(hhcd, chnum);
                        return;

                        return;
#if defined(MAX_NOTREADY_RETRY)
                    } else {
                        // MAX_NOTREADY_RETRY reached, so stop trying to resend and instead wait for a timeout at a higher layer
                    }
#endif
                    break;
            }
        }
#endif
```

On a URB_DONE they (MBED) transmit the next packet, until all packets are done. they are doing this via `td`

They also handle the NAK (URB_NOTREADY) and re transmit the last packet.

This is the code that I did not think was working , so the PR33 stops the MBED code doing this packet splitting and lets the underlying STM HAL code do it.

This fixed all my problems for 64 byte packets.

But it is not working with 8 byte packets here.

KurtE, 2023-11-02:
Thanks, as I mentioned, I think I brought in your changes... But might have messed up some... I also just now synced up the current GITHUB code changes, and then did a Winmerge to the one in libraries folder.   Will try again.

My gut tells me that a lot of these NAKs and data lost with current code is due to them doing the one packet at a time.  An in between the time they process the one packet and issue new request for the next one,  it may have lost one or more packets and then things get confused.

AndrewCapon, 2023-11-02:
Yes I think there is definitely a disconnect there, that code gets called every 20us. For the NAKs the existing code is terribly wrong it just spams the HAL submitrequest.

but the bad news, even with letting the HAL deal with the full size and some more common sense NAK code (retry, wait 10 ms before another retry) I’m still getting missed data, and I’m guessing it is the data at the start. Need to check.

I have dealt with the stm32 HAL device code before and that was pretty strange, maybe the host side us similar :)

thanks for the pic about how you gave the usb connected, I would not gave thought of doing that!

KurtE, 2023-11-03:
@AndrewCapon, For some reason, I did some more editing of my copy of the library with your changes.  And AFAIK, I did not change anything related to these changes, but now in several cases it is now  reading in the FTDI device descriptor (8 byte control endpoint)...  Also was able to read some Teensy boards configuration descriptors, which was failing.  

It is still failing to read in the Arduino UNO R4 MINIMA, will investigate this some more... Maybe tomorrow or the next day or so.

I am now curious, about when it connected up to the Teensy Micromod, I think it connected at USB FS (64 byte packets).  The Teensy also support HS (512 byte packets).  I think the USB on these STM boards support HS?  Wondering if the library supports it...  Maybe time to investigate.

AndrewCapon, 2023-11-04:
Morning @KurtE 

I have made a change that may be useful, I won't add it to the PR as I have loads of hacked stuff I am working on.

in `USBHALHost_STM.cpp` change the line 108 from

```
#if !ARC_USB_FULL_SIZE
```

to

```
#if ARC_USB_FULL_SIZE

        if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) 
        {
          static uint32_t uRetry = 0;
          if(urb_state == URB_NOTREADY)
          {
            if(uRetry == 0)
            {
              // Submit the same request again, because the device wasn't ready to accept the last one
              // should use a proper timer here but as we seem to be called every 20us just use that
              // to only re-request every 10ms
              length = td->size;
              HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
              HAL_HCD_EnableInt(hhcd, chnum);
              uRetry = 500;
            }
            else
              uRetry--;
          }
          else
            uRetry = 0;
        }
#else
```

This should help with any NAKs which the underlying HAL code doesn't handle itself.

With the problem device I have here though I am still getting missed data even though all the data is being received into the USB fifo it is not ending up in the correct place, will be looking into this today. I'm getting close I think!

AndrewCapon, 2023-11-04:
So some success:
```
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1215]SETUP PACKET: 
80 6 0 1 0 0 12 0 
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1244]CONTROL setup stage USB_TYPE_IDLE
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1274]CONTROL READ stage USB_TYPE_IDLE
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1282]CONTROL READ SUCCESS [18 bytes transferred]
12 01 10 01 00 00 00 08 56 22 10 20 00 00 01 02 00 01 

[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1315]CONTROL ack stage USB_TYPE_IDLE
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:966]CLASS: 00         VID: 2256       PID: 2010
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1197]----- CONTROL READ [dev: 0x2401078c - hub: 0 - port: 1] ------
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1211]Control transfer on device: 1

[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1215]SETUP PACKET: 
80 6 0 2 0 0 9 0 
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1244]CONTROL setup stage USB_TYPE_IDLE
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1274]CONTROL READ stage USB_TYPE_IDLE
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1282]CONTROL READ SUCCESS [9 bytes transferred]
09 02 65 00 02 01 00 C0 32 

[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1315]CONTROL ack stage USB_TYPE_IDLE
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:882]TOTAL_LENGTH: 101         NUM_INTERF: 2
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1197]----- CONTROL READ [dev: 0x2401078c - hub: 0 - port: 1] ------
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1211]Control transfer on device: 1

[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1215]SETUP PACKET: 
80 6 0 2 0 0 65 0 
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1244]CONTROL setup stage USB_TYPE_IDLE
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1274]CONTROL READ stage USB_TYPE_IDLE
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1282]CONTROL READ SUCCESS [101 bytes transferred]
09 02 65 00 02 01 00 C0 32 09 04 00 00 00 01 01 00 00 09 24 01 00 01 09 00 01 01 09 04 01 00 02 01 03 00 00 07 24 01 00 01 41 00 06 24 02 01 01 00 06 24 02 02 02 00 09 24 03 01 03 01 02 01 00 09 24 03 02 04 01 01 01 00 09 05 01 02 40 00 00 00 00 05 25 01 01 01 09 05 82 02 40 00 00 00 00 05 25 01 01 03 

[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1315]CONTROL ack stage USB_TYPE_IDLE
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:978]CONFIGURATION DESCRIPTOR:

09 02 65 00 02 01 00 C0 32 09 04 00 00 00 01 01 00 00 09 24 01 00 01 09 00 01 01 09 04 01 00 02 01 03 00 00 07 24 01 00 01 41 00 06 24 02 01 01 00 06 24 02 02 02 00 09 24 03 01 03 01 02 01 00 09 24 03 02 04 01 01 01 00 09 05 01 02 40 00 00 00 00 05 25 01 01 01 09 05 82 02 40 00 00 00 00 05 25 01 01 03 

[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1030]dev: 0x2401078c has 2 intf
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1039]ADD INTF 1 on device 0x2401078c: class: 1, subclass: 3, proto: 0
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:614]USBEndpoint created (0x2401017c): type: 2, dir: 1, size: 64, addr: 1, state: USB_TYPE_IDLE
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1058]ADD USBEndpoint 0x2401017c, on interf 1 on device 0x2401078c
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:614]USBEndpoint created (0x24010214): type: 2, dir: 2, size: 64, addr: 2, state: USB_TYPE_IDLE
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1058]ADD USBEndpoint 0x24010214, on interf 1 on device 0x2401078c
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:991]Set configuration 1 on dev: 0x2401078c
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1197]----- CONTROL WRITE [dev: 0x2401078c - hub: 0 - port: 1] ------
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1211]Control transfer on device: 1

[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1215]SETUP PACKET: 
0 9 1 0 0 0 0 0 
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1244]CONTROL setup stage USB_TYPE_IDLE
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1315]CONTROL ack stage USB_TYPE_IDLE
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1004]dev 0x2401078c is enumerated

[USB_INFO: lib/Arduino_USBHostMbed5/src/USBHostMIDI/USBHostMIDI.cpp:70]New MIDI device: VID:2256 PID:2010 [dev: 0x2401078c - intf: 1]
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.h:170]register driver for dev: 0x2401078c on intf: 1
[USB_TRANSFER: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1108]----- BULK READ [dev: 0x2401078c - MIDI - hub: 0 - port: 1 - addr: 1 - ep: 82]------
```

AndrewCapon, 2023-11-04:
I am thinking though that this is more luck than judgement, I think there is another timing issue I have not quite got yet.

here is the code to use rather than the previously posted code, it might help.

I'm still investigating what exactly is going on though because this still failed sometimes.

```
#if ARC_USB_FULL_SIZE

        if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) 
        {
          static constexpr uint32_t uPreCount = 1;
          static constexpr uint32_t uPostCount = 500;
          
          static uint32_t uNotReadyCount = 0;
          
          if(urb_state == URB_NOTREADY)
          {
            volatile uint32_t transferred = HAL_HCD_HC_GetXferCount(hhcd, chnum);

            uNotReadyCount++;

            if(uNotReadyCount == uPreCount)
            {
              // Submit the same request again, because the device wasn't ready to accept the last one
              // we need to be aware of any data that has already been transferred as it wont be again by the look of it.
              td->currBufPtr += transferred;
              td->size -= transferred;
              length = td->size;
              HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
              HAL_HCD_EnableInt(hhcd, chnum);
            }
            else if (uNotReadyCount == uPostCount)
              uNotReadyCount = 0;
          }
          else
            uNotReadyCount = 0;
        }
#else
```

AndrewCapon, 2023-11-04:
I have found a way to make it work all the time and its not good news.

```
#if ARC_USB_FULL_SIZE

        if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) 
        {
          delayMicroseconds(10);
          static constexpr uint32_t uPreCount = 1;
          static constexpr uint32_t uPostCount = 500;
          
          static uint32_t uNotReadyCount = 0;
          
          if(urb_state == URB_NOTREADY)
          {
            volatile uint32_t transferred = HAL_HCD_HC_GetXferCount(hhcd, chnum);

            uNotReadyCount++;

            if(uNotReadyCount == uPreCount)
            {
              // Submit the same request again, because the device wasn't ready to accept the last one
              // we need to be aware of any data that has already been transferred as it wont be again by the look of it.
              td->currBufPtr += transferred;
              td->size -= transferred;
              length = td->size;
              HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
              HAL_HCD_EnableInt(hhcd, chnum);
            }
            else if (uNotReadyCount == uPostCount)
              uNotReadyCount = 0;
          }
          else
            uNotReadyCount = 0;
        }
#else
```

That 10us delay is needed to synchronise the transferred count from `HAL_HCD_HC_GetXferCount(hhcd, chnum);` to this code running in the HAL isr.

Not good news!

AndrewCapon, 2023-11-04:
Right, last post for today.

It seems that you can get NAKs via URB_NOTREADY even though the data is being received, and you generally do not need to resend, just wait. But the first one you get after an URB_IDLE "maybe" needs a resend, this resend seems to need size and location adjusting by the number of bytes already received.

Now this is just on this 8 byte midi device here, and sounds a bit mad to be honest.

This code works 100% of the time here with this device:
```
void HAL_HCD_HC_NotifyURBChange_Callback(HCD_HandleTypeDef *hhcd, uint8_t chnum, HCD_URBStateTypeDef urb_state)
{
    USBHALHost_Private_t *priv = (USBHALHost_Private_t *)(hhcd->pData);
    USBHALHost *obj = priv->inst;
    void (USBHALHost::*func)(volatile uint32_t addr) = priv->transferCompleted;

    uint32_t addr = priv->addr[chnum];
    uint32_t max_size = HAL_HCD_HC_GetMaxPacket(hhcd, chnum);
    uint32_t type = HAL_HCD_HC_GetType(hhcd, chnum);
    uint32_t dir = HAL_HCD_HC_GetDirection(hhcd, chnum);
    

    uint32_t length;
    if ((addr != 0)) {
        HCTD *td = (HCTD *)addr;

#if ARC_USB_FULL_SIZE

        if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) 
        {
          
          if(urb_state == URB_NOTREADY)
          {
            volatile uint32_t transferred = HAL_HCD_HC_GetXferCount(hhcd, chnum);

            if(td->retry == 0)
            {
              // Submit the same request again, because the device wasn't ready to accept the last one
              // we need to be aware of any data that has already been transferred as it wont be again by the look of it.
              // Also only do this once until if (td->state == USB_TYPE_IDLE) resets it below
              td->currBufPtr += transferred;
              td->size -= transferred;
              td->retry = 1;
              length = td->size;

              HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
              HAL_HCD_EnableInt(hhcd, chnum);
            }
          }
        }
#else
        if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) {
            switch (urb_state) {
                case URB_DONE:
#if defined(MAX_NOTREADY_RETRY)
                    td->retry = 0;
#endif
                    if (td->size >  max_size) {
                        /*  enqueue  another request */
                        td->currBufPtr += max_size;
                        td->size -= max_size;
                        length = td->size <= max_size ? td->size : max_size;
                        HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
                        HAL_HCD_EnableInt(hhcd, chnum);
                        return;
                    }
                    break;
                case  URB_NOTREADY:
#if defined(MAX_NOTREADY_RETRY)
                    if (td->retry < MAX_NOTREADY_RETRY) {
                        td->retry++;
#endif
                        // Submit the same request again, because the device wasn't ready to accept the last one
                        length = td->size <= max_size ? td->size : max_size;
                        HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
                        HAL_HCD_EnableInt(hhcd, chnum);
                        return;

                        return;
#if defined(MAX_NOTREADY_RETRY)
                    } else {
                        // MAX_NOTREADY_RETRY reached, so stop trying to resend and instead wait for a timeout at a higher layer
                    }
#endif
                    break;
            }
        }
#endif
        if ((type == EP_TYPE_INTR)) {
            /*  reply a packet of length NULL, this will be analyze in call back
             *  for mouse or hub */
            td->state = USB_TYPE_IDLE ;
            HAL_HCD_DisableInt(hhcd, chnum);

        } else {
            if (urb_state == URB_DONE) {
                td->state = USB_TYPE_IDLE;
            }
            else if (urb_state == URB_ERROR) {
                // While USB_TYPE_ERROR in the endpoint state is used to activate error recovery, this value is actually never used.
                // Going here will lead to a timeout at a higher layer, because of ep_queue.get() timeout, which will activate error
                // recovery indirectly.
                td->state = USB_TYPE_ERROR;
            } else {
                td->state = USB_TYPE_PROCESSING;
            }
        }
        if (td->state == USB_TYPE_IDLE) {
            td->retry = 0;
            td->currBufPtr += HAL_HCD_HC_GetXferCount(hhcd, chnum);
            (obj->*func)(addr);
        }
    } else {
        if (urb_state != 0) {
            //USB_DBG_EVENT("spurious %d %d", chnum, urb_state);
        }
    }
}
```

KurtE, 2023-11-04:
Sounds a bit complicated... But will give it a try.

Wondering if they maybe had some hardware issue, or maybe code setting things in the wrong order leaving some timing window, where the full state is not ready...

AndrewCapon, 2023-11-04:
I think the issue was repeated `HAL_HCD_HC_SubmitRequest()` calls, who knows though!

alrvid, 2023-11-04:
> @alrvid - Not sure if you are the main person on this library?
> 

I'm in a team that works on libraries for Arduino Pro - so mainly libraries for Portenta boards, Nicla boards, and the Opta.  I've been doing work on this library because I'm maintaining another Arduino library that depends on the mass storage functionality of this one. I also jump in and fix all kinds of things outside my direct responsibilities because I enjoy tinkering with stuff and fixing all kinds of bugs, but only when I have time left after doing my main tasks.

By the way, thank you both @KurtE and @AndrewCapon for all your valuable contributions here! :-)

KurtE, 2023-11-04:
@AndrewCapon - tried your later stuff, and things so far appear to be running.  Did simple echo test to FTDI cable.  Plugged in UNO R4 MINIMA and it connected (did not before).  Now tried PL2303... 

@alrvid - Thanks for working on this stuff.  Hopefully we can get some of these changes in once more testing happens.

AndrewCapon, 2023-11-04:
@KurtE 

That's great, I was pretty sure it was only going to help with this problem device here :)

I'm going to test it all with a few more devices to see what else it may have broken and if all looks good then I will update the PR with the new code.

@alrvid 

👍

AndrewCapon, 2023-11-04:
Bad news, first 64byte packet midi device I have tested now doesn't work, arghhh

AndrewCapon, 2023-11-04:
Ok a slight rethink, this code works with all the midi devices I have just tried. (Disconnect and reconnect seems buggy but I'm guessing that is something unrelated)

The new Idea is that we basically ignore NAK/URB_NOTREADY for 10ms after any URB_IDLE.

If we are still getting them after 10ms we resend the request.

```
void HAL_HCD_HC_NotifyURBChange_Callback(HCD_HandleTypeDef *hhcd, uint8_t chnum, HCD_URBStateTypeDef urb_state)
{
    USBHALHost_Private_t *priv = (USBHALHost_Private_t *)(hhcd->pData);
    USBHALHost *obj = priv->inst;
    void (USBHALHost::*func)(volatile uint32_t addr) = priv->transferCompleted;

    uint32_t addr = priv->addr[chnum];
    uint32_t max_size = HAL_HCD_HC_GetMaxPacket(hhcd, chnum);
    uint32_t type = HAL_HCD_HC_GetType(hhcd, chnum);
    uint32_t dir = HAL_HCD_HC_GetDirection(hhcd, chnum);
    

    uint32_t length;
    if ((addr != 0)) {
        HCTD *td = (HCTD *)addr;

#if ARC_USB_FULL_SIZE
        constexpr uint32_t uRetryCount = 10000/20; // 10 ms (TODO: should be done with timer, investigate)
        if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) 
        {
          td->retry++;

          if(urb_state == URB_NOTREADY)
          {
            volatile uint32_t transferred = HAL_HCD_HC_GetXferCount(hhcd, chnum);

            LogicUint7(0x40 + transferred);

            if((td->retry > uRetryCount) || (td->size==0))
            {
              // Submit the same request again, because the device wasn't ready to accept the last one
              // we need to be aware of any data that has already been transferred as it wont be again by the look of it.
              // Also only do this once until if (td->state == USB_TYPE_IDLE) resets it below
              td->currBufPtr += transferred;
              td->size -= transferred;
              td->retry = 0;
              length = td->size;

              HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
              HAL_HCD_EnableInt(hhcd, chnum);
            }
          }
        }
#else
        if ((type == EP_TYPE_BULK) || (type == EP_TYPE_CTRL)) {
            switch (urb_state) {
                case URB_DONE:
#if defined(MAX_NOTREADY_RETRY)
                    td->retry = 0;
#endif
                    if (td->size >  max_size) {
                        /*  enqueue  another request */
                        td->currBufPtr += max_size;
                        td->size -= max_size;
                        length = td->size <= max_size ? td->size : max_size;
                        HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
                        HAL_HCD_EnableInt(hhcd, chnum);
                        return;
                    }
                    break;
                case  URB_NOTREADY:
#if defined(MAX_NOTREADY_RETRY)
                    if (td->retry < MAX_NOTREADY_RETRY) {
                        td->retry++;
#endif
                        // Submit the same request again, because the device wasn't ready to accept the last one
                        length = td->size <= max_size ? td->size : max_size;
                        HAL_HCD_HC_SubmitRequest(hhcd, chnum, dir, type, !td->setup, (uint8_t *) td->currBufPtr, length, 0);
                        HAL_HCD_EnableInt(hhcd, chnum);
                        return;

                        return;
#if defined(MAX_NOTREADY_RETRY)
                    } else {
                        // MAX_NOTREADY_RETRY reached, so stop trying to resend and instead wait for a timeout at a higher layer
                    }
#endif
                    break;
            }
        }
#endif
        if ((type == EP_TYPE_INTR)) {
            /*  reply a packet of length NULL, this will be analyze in call back
             *  for mouse or hub */
            td->state = USB_TYPE_IDLE ;
            HAL_HCD_DisableInt(hhcd, chnum);

        } else {
            if (urb_state == URB_DONE) {
                td->state = USB_TYPE_IDLE;
            }
            else if (urb_state == URB_ERROR) {
                // While USB_TYPE_ERROR in the endpoint state is used to activate error recovery, this value is actually never used.
                // Going here will lead to a timeout at a higher layer, because of ep_queue.get() timeout, which will activate error
                // recovery indirectly.
                td->state = USB_TYPE_ERROR;
            } else {
                td->state = USB_TYPE_PROCESSING;
            }
        }
        if (td->state == USB_TYPE_IDLE) {
#if ARC_USB_FULL_SIZE
            td->retry = 0;
#endif
            td->currBufPtr += HAL_HCD_HC_GetXferCount(hhcd, chnum);
            (obj->*func)(addr);
        }
    } else {
        if (urb_state != 0) {
            //USB_DBG_EVENT("spurious %d %d", chnum, urb_state);
        }
    }
}
```

AndrewCapon, 2023-11-04:
Actually seems to work through a hub as well!

As long as you only plug one device in that is.

Two devices and red lights of doom, will investigate tomorrow. I need to get hubs working to have multiple controllers...

KurtE, 2023-11-07:
@AndrewCapon @mjs513 - 
Sorry just tried your latest stuff... 

I am getting several warnings, plus an error on the line:
`          LogicUint7(0x40 + transferred);`
LogicUint7 is not defined anywhere...

AndrewCapon, 2023-11-07:
Hi @KurtE 

Sorry that's some of my test code that I didn't remove, just delete that line.

I will update the PR today.

I have also managed to get multiple devices going, there is an issue with the MBED Hub code though, once a device is connected the HUB code seems to stop working and it doesn't recognise new devices. Will look into that a bit today...

KurtE, 2023-11-17:
@AndrewCapon @mjs513 - Not sure if this is best place to mention.  I think I still have one problem child with reading in descriptors.  In this case it is a HID descriptor for a Wacom tablet.   This descriptor is 759 bytes long. 

It errors out on the read on the GIGA.  I verified it reads fine on Teensy.   And the parser there properly decoded it.
On GIGA:
```

*** starting Device information sketch ***
No USB host device connected
Device:0x2400c72c
VID: 56A, PID: 374

----------------------------------------
Device Descriptor:
  bcdUSB: 512
  bDeviceClass: 0
  bDeviceSubClass: 0
  bDeviceProtocol: 0
  bMaxPacketSize: 64
  idVendor: 0x56a
  idProduct: 0x374
  bcdDevice: 273
  iManufacturer: 1
  iProduct: 2
  iSerialNumber: 3
  bNumConfigurations: 1
Manufacturer: Wacom Co.,Ltd.
Product: Intuos S
Serial Number: 1GE00R2010013
Speed: 0

----------------------------------------
Size of configuration Descriptor: 34
Configuration Descriptor
2400E8E8 - 09 02 22 00 01 01 00 80  FA 09 04 00 00 01 03 00  : .."..... ........
2400E8F8 - 00 00 09 21 10 01 00 01  22 F7 02 07 05 81 03 40  : ...!.... "......@
2400E908 - 00 01                                             : ..
Config:
 wTotalLength: 34
 bNumInterfaces: 1
 bConfigurationValue: 1
 iConfiguration: 0
 bmAttributes: 128
 bMaxPower: 250
****************************************
** Interface level **
  bInterfaceNumber: 0
  bAlternateSetting: 0
  Number of endpoints: 1
  bInterfaceClass: 3
  bInterfaceSubClass: 0
    HID
  bInterfaceProtocol: 0
    None
  iInterface: 0
  HID Descriptor size: 759
  Endpoint: 81 In
    Attrributes: 3 Interrupt
    Size: 64
    Interval: 1
>>>>> USBDumperDevice::getHIDDesc(0) called <<<<< 
*** Failed to read in HID Report Descriptor ***
parseInterface nb:0
 bInterfaceClass = 3
 bInterfaceSubClass = 0
    HID
 bProtocol = 0
    None
useEndpoint(0, 3, 2)
New Debug device: VID:056a PID:0374 [dev: 0x2400c72c - intf: 0]
USB host device(56a:374) connected
[... 376 lines cut ...]
  09 01	// Usage(1) -
  96 01 00	// Report Count(1)
  B1 02	// Feature(2)	// (Data, Variable, Absolute)
  85 E1	// Report ID(e1)
  09 01	// Usage(1) -
  96 02 00	// Report Count(2)
  B1 02	// Feature(2)	// (Data, Variable, Absolute)
  85 E2	// Report ID(e2)
  09 01	// Usage(1) -
  96 02 00	// Report Count(2)
  B1 02	// Feature(2)	// (Data, Variable, Absolute)
  85 E3	// Report ID(e3)
  09 01	// Usage(1) -
  96 02 00	// Report Count(2)
  B1 02	// Feature(2)	// (Data, Variable, Absolute)
  85 E4	// Report ID(e4)
  09 01	// Usage(1) -
  96 FF 01	// Report Count(1ff)
  B1 02	// Feature(2)	// (Data, Variable, Absolute)
  C0	// End Collection
*** HID Device hdc1 56a: 374 - connected ***
  manufacturer: Wacom Co.,Ltd.
  product: Intuos S
  Serial: 1GE00R2010013
```

AndrewCapon, 2023-11-17:
Hi @KurtE 

Can you set the debug fully on in dbg.h:

```
#define DEBUG 4 /*INFO,ERR,WARN*/
#define DEBUG_TRANSFER 1
#define DEBUG_EP_STATE 1
#define DEBUG_EVENT 1
```

And post the log here.

KurtE, 2023-11-17:
```
*** starting Device information sketch ***
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:89]*** DEVICE_CONNECTED_EVENT: 0 1 0 0x0

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:120]HPRT Speed: 1: Low speed? 0

[USB_EVENT: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:130]new device connected: 0x2400c72c

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBDeviceConnected.cpp:88]init dev: 0x2400c72c
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:629]USBEndpoint created (0x2400c084): type: 0, dir: 1, size: 8, addr: 0, state: USB_TYPE_IDLE
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]0x2400c084 0 1 2 1

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:91](SA)HAL_HCD_HC_Init(0x2400131c 0 0 0 1 0 8)

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:642]Resetting hub 0, port 1

No USB host device connected
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:166]usb_thread read device descriptor on dev: 0x2400c72c

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1210]----- CONTROL READ [dev: 0x2400c72c - hub: 0 - port: 1] ------
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:99]** USBEndpoint::setSpeed(0x2400c084, 0) **

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]0x2400c084 0 1 2 1

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:91](SA)HAL_HCD_HC_Init(0x2400131c 0 0 0 1 0 8)

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1224]Control transfer on device: 0

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1228]SETUP PACKET: 
80 6 0 1 0 0 8 0 
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1256]CONTROL setup stage USB_TYPE_IDLE
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1285]CONTROL READ stage USB_TYPE_IDLE
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1293]CONTROL READ SUCCESS [8 bytes transferred]
12 01 00 02 00 00 00 40 

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1326]CONTROL ack stage USB_TYPE_IDLE
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1210]----- CONTROL WRITE [dev: 0x2400c72c - hub: 0 - port: 1] ------
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:99]** USBEndpoint::setSpeed(0x2400c084, 0) **

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]0x2400c084 0 1 2 1

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:91](SA)HAL_HCD_HC_Init(0x2400131c 0 0 0 1 0 64)

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1224]Control transfer on device: 0

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1228]SETUP PACKET: 
0 5 1 0 0 0 0 0 
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1256]CONTROL setup stage USB_TYPE_IDLE
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1326]CONTROL ack stage USB_TYPE_IDLE
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:185]Address of 0x2400c72c: 1
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1210]----- CONTROL READ [dev: 0x2400c72c - hub: 0 - port: 1] ------
[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:99]** USBEndpoint::setSpeed(0x2400c084, 0) **

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]0x2400c084 0 1 2 1

[USB_DBG: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:91](SA)HAL_HCD_HC_Init(0x2400131c 0 0 1 1 0 64)

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1224]Control transfer on device: 1

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1228]SETUP PACKET: 
80 6 0 1 0 0 8 0 
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1256]CONTROL setup stage USB_TYPE_IDLE
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1285]CONTROL READ stage USB_TYPE_IDLE
[... cut ...]
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1285]CONTROL READ stage USB_TYPE_IDLE
[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1293]CONTROL READ SUCCESS [28 bytes transferred]
1C 03 31 00 47 00 45 00 30 00 30 00 52 00 32 00 30 00 31 00 30 00 30 00 31 00 33 00 

[USB_TRANSFER: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\USBHost\USBHost.cpp:1326]CONTROL ack stage USB_TYPE_IDLE
Serial Number: 1GE00R2010013
```

AndrewCapon, 2023-11-17:
Hi @KurtE 

Can you try changing in `void HAL_HCD_HC_NotifyURBChange_Callback(HCD_HandleTypeDef *pHcd, uint8_t uChannel, HCD_URBStateTypeDef urbState)` in `USBHALHost_STM.cpp`

```
constexpr uint32_t uRetryCount = 10;
```

Try making that retry count larger, maybe 20, then 50, then 100 to see if that makes it work.

Really this retry count should be based on a timer as `HAL_HCD_HC_NotifyURBChange_Callback` can be called differently for different devices, that 10 was a arbitrary figure I came up with here which equates to about 4ms from the number of times it was called here, this may be different for different devices.

KurtE, 2023-11-17:
Thanks, will try and let you know soon.

KurtE, 2023-11-17:
Did not notice any difference.  I think I may need to debug more... I am not seeing the tablet responding to the request...
Maybe it does not like something with it...

KurtE, 2023-11-17:
Here is a query for the HID Descriptor done by Teensy.  
[image]
A more closeup showing the setup packet, and the ones up to when the first Input data started coming bacK
[image]
And the color on top shows my HLA analyzer output for the setup and in...
[image]

However on the GIGA, we see at the request time.
[image]

And the next actual frame with any information is about a second later:
[image]
[image]

Almost like the IN endpoint is not active...  Wondering if maybe the register to setup for transfer size is computing correctly tne number of return packets and allowing it.

KurtE, 2023-11-18:
I think there is a hint in the request failing:
[image]
Note the last thing on the right.  The IN packet ended up in STALED state... 

And I noticed in:
`void HAL_HCD_HC_NotifyURBChange_Callback(HCD_HandleTypeDef *pHcd, uint8_t uChannel, HCD_URBStateTypeDef urbState)
`
The comments:
```
//  - URB_STALL    = a stall response from the device - but it is never handled by the library and will end up
//                   as a timeout at a higher layer, because of an ep_queue.get() timeout, which will activate
//                   error recovery indirectly

```

AndrewCapon, 2023-11-18:
What's the time between the NAK and the STALL?

Also yesterday I was a bit confused, these HID requests are on the Control Endpoint not the Interrupt Endpoint?

Also as a quick test can you set `uRetryCount` to 0 and give that a go. (instant retry on NAK)
And set to 0xffffffff as well. (No retries at all)

KurtE, 2023-11-18:
[image]

Will try some of the other retry and see what that does.  It may be that the tablet gets anxious and gives up quickly as it has 12 packets to send back.

AndrewCapon, 2023-11-18:
I wonder if it is something with the lower level NAK handling.

From what I understand from the traces the teensy is sending multiple NAKs while it works it out and then the data, looks good.

The GIGA is getting a single NAK and then a STALL.

Does that sound about right?

KurtE, 2023-11-18:
[image]

Same as before, except it sends it 1 more time, which still returns stalled. 
And then back to only SOF packets every ms for about a second until I issue the next request

I should mention that it does work on some devices, here is keyboard USB LS 8 byte control packets
[image]

And a different tablet USB HS - has two report descriptors:
First is only 18 bytes
[image]

Second one is 119 bytes:  But again 8 byte control endpoint so lots of packets... spread over a lot of time:
[image]

On the Teensy with the tablet that fails on GIGA:
[image]

So yes 3 NAKs before it responds true.  But the time between when the request finished sending until the In attempt that starts retrieving data is about 21us

So wondering if timing?

AndrewCapon, 2023-11-18:
Can you notice any differences between the USB data going from the teensy/giga to the tablet?

I do have a Wacom tablet somewhere but I have no idea where it is, I will attempt to spot it tomorrow...

KurtE, 2023-11-18:
As I mentioned, I wonder if some of this is about timing and the usage of the frames.  That is I think the current code will check for the IN data something like once per frame.  
It was hard for me to see exactly where the SOF frames are in some of these captures, so I spent a little time this morning hacking up a new HLA that put a "*" up at each SOF...

So for example with the last tablet I mentioned, (HUION) with the 119 byte HID descriptor which takes 15 data packets.

Again looks like we only ask for data once per 1ms frame:
[image]
In this case the device is really slow so maybe does not matter as much:

On the Micromod:
The USB Frames are sliced up into more parts.  Probably depending on how many things are wanting some of the USB Band width.
But for example the slow responding device for the HID descriptor shown earlier,  Here is showing the frame with the request and start of the next frame:
[image]

And each of those sections within those portions are asking multiple times sometimes 15 times or so.
[image]

But this is the control end point... For later on when waiting for input it is just asking once per frame.

AndrewCapon, 2023-11-18:
Doesn't It depend on the device, so if it is using frames it is 1ms, if it is using microframes it is 125us.

KurtE, 2023-11-18:
I am also not an expert on some of this stuff, I have browsed through the USB documents some, such that I have some basic
understandings.

I have also done a reasonable amount of work on the sort of layers above this low level stuff.  PaulStoffregen does most/all of this lower stuff on the Teensy.

One of the websites I often go to with USB questions is:
https://beyondlogic.org/usbnutshell/usb4.shtml#Control

If you go to the bottom of the above page, it talks about some of the bandwidth management stuff...

AndrewCapon, 2023-11-19:
That is for ISO and INTERRUPT endpoints though.

When dealing with BULK and CONTROL I am pretty sure the transfer throughput is based on (Frames & packet size) or (Microframes and (packet size).

So if the device is using Frames and an 8 byte packet and you are transferring 64 bytes it will take 8ms at best. 8 bytes per frame (1ms)

If the device is using Microframes and an 8 byte packet and you are transferring 64 bytes then it will take 1ms at best. 8 bytes per microframe (125us)

AndrewCapon, 2023-11-19:
I found my cheap Wacom, unfortunately it works though:
```
*** starting Device information sketch ***
No USB[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBDeviceConnected.cpp:88]init dev: 0x2400f84c
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:634]USBEndpoint created (0x2400f1a4): type: 0, dir: 1, size: 8, addr: 0, state: USB_TYPE_IDLE
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:647]Resetting hub 0, port 1

 host device connected
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:162]usb_thread read device descriptor on dev: 0x2400f84c

[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:181]Address of 0x2400f84c: 1
[USB_INFO: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:206]New device connected: 0x2400f84c [hub: 0 - port: 1]
Device:0x2400f84c
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:963]dev: 0x2400f84c nb_intf: 0
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:964]dev: 0x2400f84c nb_intf_attached: 0
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:970]Enumerate dev: 0x2400f84c
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:986]CLASS: 00         VID: 056A       PID: 00D2
VID: 56A, PID: d2
Speed: 0
Size of configuration Descriptor: 59
[USB_DBG: src/USBDumperDevice.cpp:406]TOTAL_LENGTH: 59   NUM_INTERF: 2
 bNumInterfaces: 2
 bConfigurationValue: 1
 iConfiguration: 0
 bmAttributes: 128
 bMaxPower: 49
  bInterfaceNumber: 0
  bAlternateSetting: 0
  Number of endpoints: 1
  bInterfaceClass: 3
  bInterfaceSubClass: 1
  bInterfaceProtocol: 2
  iInterface: 0
  HID Descriptor size: 176
81    Attrributes: 3    Size: 9
    Interval: 4
>>>>> USBDumperDevice::getHIDDesc(0) called <<<<< 

HID Report Descriptor (0x24011228) size: 176
  05 01 // Usage Page(1) -   09 02      // Usage(2) -  A1 01    // Collection(1) top Usage(10000)  85 01        // Report ID(1)  09 01  // Usage(1) -  A1 00    // Collection(0)  05 09 // Usage Page(9) -   19 01 // Usage Minimum(1) -  (BUTTON 1)  29 05        // Usage Maximum(5) -  (BUTTON 5)  15 00        // Logical Minimum(0)  25 01    // Logical maximum(1)  95 05    // Report Count(5)  75 01       // Report Size(1)  81 02   // Input(2)     // (  95 01     // Report Count(1)  75 03       // Report Size(3)  81 01        // Input(1)     // (  05 01     // Usage Page(1) -   09 30      // Usage(30) -  09 31      // Usage(31) -  15 81   // Logical Minimum(81)  25 7F   // Logical maximum(7f)  75 08   // Report Size(8)  95 02        // Report Count(2)  81 06       // Input(6)     // (  C0  C0  05 0D     // Usage Page(d) -   09 01 // Usage(1) -  A1 01    // Collection(1) top Usage(d0000)  85 02        // Report ID(2)  A1 00  // Collection(0)  06 00 FF      // Usage Page(ff00) -   09 01   // Usage(1) -  15 00       // Logical Minimum(0)  26 FF 00 // Logical maximum(ff)  75 08   // Report Size(8)  95 08        // Report Count(8)  81 02       // Input(2)     // (  C0  09 01 // Usage(1) -  85 02    // Report ID(2)  95 01     // Report Count(1)  B1 02       // Feature(2)   // (  09 01     // Usage(1) -  85 03    // Report ID(3)  95 01  // Report Count(1)  B1 02       // Feature(2)   // (  09 01     // Usage(1) -  85 04       // Report ID(4)  95 01  // Report Count(1)  B1 02       // Feature(2)   // (  09 01     // Usage(1) -  85 05    // Report ID(5)  95 01  // Report Count(1)  B1 02       // Feature(2)   // (  09 01        // Usage(1) -  85 10    // Report ID(10)  95 02 // Report Count(2)  B1 02       // Feature(2)   // (  09 01     // Usage(1) -  85 11    // Report ID(11)  95 10 // Report Count(10)  B1 02// Feature(2)    // (  09 01     // Usage(1) -  85 13    // Report ID(13)  95 01 // Report Count(1)  B1 02       // Feature(2)   // (  09 01     // Usage(1) -  85 20    // Report ID(20)  95 01 // Report Count(1)  B1 02  // Feature(2)   // (  09 01     // Usage(1) -  85 21    // Report ID(21)  95 01 // Report Count(1)  B1 02       // Feature(2)   // (  09 01     // Usage(1) -  85 06    // Report ID(6)  95 01     // Report Count(1)  B1 02       // Feature(2)   // (  09 01     // Usage(1) -  85 07    // Report ID(7)  95 01  // Report Count(1)  B1 02       // Feature(2)   // (  09 01     // Usage(1) -  85 14       // Report ID(14)  95 01 // Report Count(1)  B1 02       // Feature(2)   // (  C0  bInterfaceNumber: 1
  bAlternateSetting: 0
  Number of endpoints: 1
  bInterfaceClass: 3
  bInterfaceSubClass: 0
  bInterfaceProtocol: 0
  iInterface: 0
  HID Descriptor size: 75
82    Attrributes: 3    Size: 64
    Interval: 4
>>>>> USBDumperDevice::getHIDDesc(1) called <<<<< 

HID Report Descriptor (0x24011290) size: 75
  06 00 FF      // Usage Page(ff00) -   09 01   // Usage(1) -  A1 01    // Collection(1) top Usage(ff000000)  85 02     // Report ID(2)  05 0D  // Usage Page(d) -   09 22      // Usage(22) -  A1 00   // Collection(0)  06 00 FF // Usage Page(ff00) -   09 01   // Usage(1) -  15 00    // Logical Minimum(0)  26 FF 00 // Logical maximum(ff)  75 08   // Report Size(8)  95 02        // Report Count(2)  81 02 // Input(2)      // (  05 01     // Usage Page(1) -   09 30      // Usage(30) -  35 00   // Physical Minimum(0)  46 E0 2E        // Physical Maximum(2ee0)  26 E0 01     // Logical maximum(1e0)  75 10  // Report Size(10)  95 01  // Report Count(1)  81 02       // Input(2)     // (  09 31     // Usage(31) -  46 40 1F        // Physical Maximum(1f40)  26 40 01     // Logical maximum(140)  81 02  // Input(2)// (  06 00 FF  // Usage Page(ff00) -   09 01   // Usage(1) -  26 FF 00 // Logical maximum [... cut ...]
 bInterfaceClass = 3
 bInterfaceSubClass = 0
 bProtocol = 0
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1012]Set configuration 1 on dev: 0x2400f84c
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.cpp:1025]dev 0x2400f84c is enumerated

New Debug device: VID:056a PID:00d2 [dev: 0x2400f84c - intf: 0]
[USB_DBG: lib/Arduino_USBHostMbed5/src/USBHost/USBHost.h:172]register driver for dev: 0x2400f84c on intf: 0
USB host device(56a:d2) connected
Manufacturer: Wacom Co.,Ltd.
Product: CTH-461
```

AndrewCapon, 2023-11-19:
Just tried a HUION one as well, also worked as well, even smaller descriptor size though.

AndrewCapon, 2023-11-19:
I'm running through in the debugger with a size of 759

Looking in `stm32h7xx_ii_usb.c"
```

  /* Compute the expected number of packets associated to the transfer */
  if (hc->xfer_len > 0U)
  {
    num_packets = (uint16_t)((hc->xfer_len + hc->max_packet - 1U) / hc->max_packet);

    if (num_packets > max_hc_pkt_count)
    {
      num_packets = max_hc_pkt_count;
      hc->XferSize = (uint32_t)num_packets * hc->max_packet;
    }
  }
  else
  {
    num_packets = 1U;
  }
```
max_hc_pkt_count is 256 so that's not the problem, it calculates (8 byte packets) that we need 95 packets correctly.

Then:

```
/*
   * For IN channel HCTSIZ.XferSize is expected to be an integer multiple of
   * max_packet size.
   */
  if (hc->ep_is_in != 0U)
  {
    hc->XferSize = (uint32_t)num_packets * hc->max_packet;
  }
  else
  {
    hc->XferSize = hc->xfer_len;
  }
```

We get a XferSize of 760 here.

So it looks like the underlying stuff is calculating the packets needed correctly.

AndrewCapon, 2023-11-19:
Hi @KurtE 

While I was looking at the HUION tablet I noticed that we were sometimes still getting USB interrupt flooding despite the NAK fixes.

So I looked into this and tracked it down to the "Channel Halted Interrupt".

I have updated https://github.com/arduino-libraries/Arduino_USBHostMbed5/pull/42 to also stop this CHH interrupt flooding.

Might be worth you giving this version a go, maybe there is an outside chance this might have something to do with your issue.

AndrewCapon, 2023-11-19:
Hold off on those recent changes, with the CHH interrupt flooding turned off we are not getting any data from the interrupt endpoint, looking into it...

AndrewCapon, 2023-11-19:
Ok fixed that issue, the CHH flooding is only disabled for the CONTROL endpoints now.

Might be worth a go to see if it sorts your issue, slim chance I would guess.

[2 more replies]