arduino-libraries/Arduino_USBHostMbed5 #43: Support Low Speed device? like many boot level keyboards?

KurtE, 2023-11-11:
There are several devices like keyboards, which run in USB Low Speed (1.5MBps) and are not properly being handled.
For example, I have a Dell keyboard plugged in.  I believe this one came with my newer computer I purchased in the last couple of years. 

If I run the KeyboardGiga example sketch from the USBHostGiga library it receives, the keystrokes and prints them out in the Serial monitor. 

If I instead run a similar test using this library it does not.
```
/*
  USBHost Keyboard test

  The circuit:
   - Arduino GIGA

  This example code is in the public domain.
*/

#include <Arduino_USBHostMbed5.h>
#include <USBHostHID/USBHostKeyboard.h>
REDIRECT_STDOUT_TO(Serial)

USBHostKeyboard kbd;

// If you are using a Portenta Machine Control uncomment the following line
// mbed::DigitalOut otg(PB_14, 0);

void setup()
{
    Serial.begin(115200);
    while (!Serial) {}

    Serial.println("Starting Keyboard test...");

    // Enable the USBHost 
    pinMode(PA_15, OUTPUT);
    digitalWrite(PA_15, HIGH);

    // if you are using a Max Carrier uncomment the following line
    // start_hub();
    kbd.attach(&kbd_key_cb);
    kbd.attach(&kbd_keycode_cb);

    while (!kbd.connect()) {
      Serial.println("No keyboard connected");        
        delay(5000);
    }
}

void loop()
{
    delay(1000);
}

void kbd_key_cb(uint8_t key) {
  Serial.print("Key pressed: ");
  Serial.print(key, HEX);
  Serial.print("(");
  if ((key >= ' ') && (key <= '~')) Serial.write(key);
  Serial.println(")");
}

void kbd_keycode_cb(uint8_t keycode, uint8_t mod) {
  Serial.print("Keycode: ");
  Serial.print(keycode, HEX);
  Serial.print(" mod: ");
  Serial.println(mod, HEX);
}
```
I added debug output to the keyboard class...  So I know it is being called.  Debug shows:
```
Starting Keyboard test...
[... 246 lines cut ...]
5.249177614 | IN | 0x0 | 0x1 |   | 0x9 0x2 0x3b 0x0 0x2 0x1 0x0 0xa0 |  
5.249328152 | IN | 0x0 | 0x1 |   | 0x32 |   |   |   |  
5.24949202 | SETUP | 0x0 | 0x1 | [GET_DESCRIPTOR - CONFIG #:0 I:0x0   L:0x3b] | 0x80 0x6 0x0 0x2 0x0 0x0 0x3b 0x0 |  
5.249817728 | IN | 0x0 | 0x1 |   | 0x9 0x2 0x3b 0x0 0x2 0x1 0x0 0xa0 |  
5.250017522 | IN | 0x0 | 0x1 |   | 0x32 0x9 0x4 0x0 0x0 0x1 0x3 0x1 |  
5.250218148 | IN | 0x0 | 0x1 |   | 0x1 0x0 0x9 0x21 0x10 0x1 0x0 0x1 |  
5.250416976 | IN | 0x0 | 0x1 |   | 0x22 0x41 0x0 0x7 0x5 0x81 0x3 0x8 |  
5.25081862 | IN | 0x0 | 0x1 |   | 0x0 0x18 0x9 0x4 0x1 0x0 0x1 0x3 |  
5.251016772 | IN | 0x0 | 0x1 |   | 0x0 0x0 0x0 0x9 0x21 0x10 0x1 0x0 |  
5.251216778 | IN | 0x0 | 0x1 |   | 0x1 0x22 0x9f 0x0 0x7 0x5 0x82 0x3 |  
5.25141592 | IN | 0x0 | 0x1 |   | 0x8 0x0 0xa |   |   |  
5.257467042 | SETUP | 0x0 | 0x1 | [SET_CONFIGURATION I:0x0   L:0x0] | 0x0 0x9 0x1 0x0 0x0 0x0 0x0 0x0 |  
5.762888028 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 |  
5.85990304 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x38 0x0 0x0 0x0 0x0 0x0 |  
5.952923144 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 |  
6.079945542 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x38 0x0 0x0 0x0 0x0 0x0 |  
6.228969228 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 |  
6.351984988 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x38 0x0 0x0 0x0 0x0 0x0 |  
6.476997196 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 |  
6.59402179 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x38 0x0 0x0 0x0 0x0 0x0 |  
6.71103967 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 |  
6.816071866 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x10 0x0 0x0 0x0 0x0 0x0 |  
7.007111472 | IN | 0x1 | 0x1 |   | 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 |  
</body>
</html>

AndrewCapon, 2023-11-11:
Without a hub this code is used:
```
void HAL_HCD_Connect_Callback(HCD_HandleTypeDef *hhcd)
{
    USBHALHost_Private_t *priv = (USBHALHost_Private_t *)(hhcd->pData);
    USBHALHost *obj = priv->inst;
    void (USBHALHost::*func)(int hub, int port, bool lowSpeed, USBHostHub * hub_parent) = priv->deviceConnected;
    (obj->*func)(0, 1, 0, NULL);
}
```

the hub code has this:
```
                status = getPortStatus(port);
                USB_DBG("[hub handler hub: %d] status port %d [hub: %p]: 0x%X", dev->getHub(), port, dev, status);

                // if connection status has changed
                if (status & C_PORT_CONNECTION) {
                    if (status & PORT_CONNECTION) {
                        USB_DBG("[hub handler hub: %d - port: %d] new device connected", dev->getHub(), port);
                        host->deviceConnected(dev->getHub() + 1, port, status & PORT_LOW_SPEED, this);
                    } else {
                        USB_DBG("[hub handler hub: %d - port: %d] device disconnected", dev->getHub(), port);
                        host->deviceDisconnected(dev->getHub() + 1, port, this, 0);
                    }

                    clearPortFeature(C_PORT_CONNECTION_FEATURE, port);
                }
...
```

getPortStatus():
```
uint32_t USBHostHub::getPortStatus(uint8_t port)
{
    uint32_t st;
    host->controlRead(  dev,
                        USB_DEVICE_TO_HOST | USB_REQUEST_TYPE_CLASS | USB_RECIPIENT_INTERFACE | USB_RECIPIENT_ENDPOINT,
                        GET_STATUS,
                        0,
                        port,
                        (uint8_t *)&st,
                        4);
    return st;
}
```

AndrewCapon, 2023-11-11:
It looks like it isn't straightforward to get the speed of the direct connection:

```
(if DP is pulled high -> FS, if DM is high -> LS).
```

I don't know hat the timing for that being true is but I'm guessing not when `HAL_HCD_Connect_Callback()` is called.

KurtE, 2023-11-11:
It looks like the other library (USBHostGiga) uses the lower level function:
```
USBH_SpeedTypeDef USBH_LL_GetSpeed(USBH_HandleTypeDef *phost)
{
  USBH_SpeedTypeDef speed = USBH_SPEED_FULL;

  switch (HAL_HCD_GetCurrentSpeed(phost->pData))
  {
  case 0 :
    speed = USBH_SPEED_HIGH;
    break;

  case 1 :
    speed = USBH_SPEED_FULL;
    break;

  case 2 :
    speed = USBH_SPEED_LOW;
    break;

  default:
   speed = USBH_SPEED_FULL;
    break;
  }
  return  speed;
}
```

The low level code then reads in the register OTG_HPRT
and looks at the field:
>Bits 18:17 PSPD[1:0]: Port speed
Indicates the speed of the device attached to this port.
01: Full speed
10: Low speed
11: Reserved
00: High speed

KurtE, 2023-11-11:
Making some progress... I am pretty sure a real solution might need to abstract some stuff...
But I am detecting a low-speed device:
```
                        } else {
                            hub_unplugged = false;
                            #if defined(TARGET_GIGA)

                            #define USBx_BASE   USB1_OTG_HS_PERIPH_BASE
                            #define USBx_HPRT0      *(__IO uint32_t *)((uint32_t)USBx_BASE + USB_OTG_HOST_PORT_BASE)
                            uint8_t pspd = ((USBx_HPRT0 & USB_OTG_HPRT_PSPD) >> 17);
                            if (pspd == 2) {
                                usb_msg->lowSpeed = true;
                            }
                            printf("HPRT Speed: %u: Low speed? %u\n", pspd, usb_msg->lowSpeed );
                            #endif
                        }

                        if (((idx!=-1) && deviceInUse[idx] ) || ((idx == -1) && hub_unplugged)) {
```

And I pugged in a LS device (the keyboard) and likewise a Serial device, that is FS... And they were detect.

So then running the Keyboard test program mentioned in the original post here.  And getting lots of Warnings output, and still now callbacks with the characters. 

```
Starting Keyboard test...
No keyboard connected
*** DEVICE_CONNECTED_EVENT: 0 1 0 0x0
HPRT Speed: 2: Low speed? 1
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]small speed device on hub not supported
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]small speed device on hub not supported
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]small speed device on hub not supported
got device
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]small speed device on hub not supported
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]small speed device on hub not supported
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]small speed device on hub not supported
intf_class: 3
intf_subclass: 1
intf_protocol: 1
intf_nb: 0
intf_class: 3
intf_subclass: 0
intf_protocol: 0
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:89]small speed device on hub not supported
ret: 0
after enumerate
```

Now to see if that can be resolved...

AndrewCapon, 2023-11-12:
That isn't a reassuring warning!

KurtE, 2023-11-12:
Still working on it, 
```
Starting Keyboard test...
No keyboard connected
*** DEVICE_CONNECTED_EVENT: 0 1 0 0x0
HPRT Speed: 2: Low speed? 1
0x2400ce0c 0 1 2 1
(SA)HAL_HCD_HC_Init(0x240012b4 0 0 0 1 0 8)
** USBEndpoint::setSpeed(0x2400ce0c, 1) **
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:87]low speed device on hub not supported
0x2400ce0c 1 1 2 2
(SA)HAL_HCD_HC_Init(0x240012b4 0 0 0 2 0 8)
** USBEndpoint::setSpeed(0x2400ce0c, 1) **
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:87]low speed device on hub not supported
0x2400ce0c 1 1 2 2
(SA)HAL_HCD_HC_Init(0x240012b4 0 0 0 2 0 8)
** USBEndpoint::setSpeed(0x2400ce0c, 1) **
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:87]low speed device on hub not supported
0x2400ce0c 1 1 2 2
(SA)HAL_HCD_HC_Init(0x240012b4 0 0 1 2 0 8)
got device
** USBEndpoint::setSpeed(0x2400ce0c, 1) **
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:87]low speed device on hub not supported
0x2400ce0c 1 1 2 2
(SA)HAL_HCD_HC_Init(0x240012b4 0 0 1 2 0 8)
** USBEndpoint::setSpeed(0x2400ce0c, 1) **
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:87]low speed device on hub not supported
0x2400ce0c 1 1 2 2
(SA)HAL_HCD_HC_Init(0x240012b4 0 0 1 2 0 8)
** USBEndpoint::setSpeed(0x2400ce0c, 1) **
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:87]low speed device on hub not supported
0x2400ce0c 1 1 2 2
(SA)HAL_HCD_HC_Init(0x240012b4 0 0 1 2 0 8)
intf_class: 3
intf_subclass: 1
intf_protocol: 1
intf_nb: 0
** USBEndpoint::setSpeed(0x2400cea4, 1) **
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:87]low speed device on hub not supported
0x2400cea4 1 1 2 2
(SA)HAL_HCD_HC_Init(0x240012b4 1 129 1 2 3 8)
** USBEndpoint::setSpeed(0x2400cea4, 1) **
intf_class: 3
intf_subclass: 0
intf_protocol: 0
** USBEndpoint::setSpeed(0x2400ce0c, 1) **
[USB_WARNING: c:\Users\kurte\Documents\Arduino\libraries\Arduino_USBHostMbed5\src\targets\TARGET_STM\USBEndpoint_STM.cpp:87]low speed device on hub not supported
0x2400ce0c 1 1 2 2
(SA)HAL_HCD_HC_Init(0x240012b4 0 0 1 2 0 8)
ret: 0
after enumerate
End of Setup
```
The calls to HAL_HCD_HC_init, are where the end points get setup and the speed is passed in.... 
0=HS 1=FS 2=LS.
I have most of them caught to pass in 2... But one more to go... Not sure if there are other things needed as well

But still no keystrokes

KurtE, 2023-11-12:
@AndrewCapon @mjs513 - Making some progress on this...  Still needs a lot of cleaning up.
I pushed up a new branch: https://github.com/KurtE/Arduino_USBHostMbed5/tree/usb_host_fixes_lowspeed
Which is based off of your usb_host_fixes.
```
/*
  USBHost Keyboard test

  The circuit:
   - Arduino GIGA

  This example code is in the public domain.
*/

#include <Arduino_USBHostMbed5.h>
#include <USBHostHID/USBHostKeyboard.h>
REDIRECT_STDOUT_TO(Serial)

USBHostKeyboard kbd;

// If you are using a Portenta Machine Control uncomment the following line
// mbed::DigitalOut otg(PB_14, 0);

void setup()
{
    Serial.begin(115200);
    while (!Serial && millis() < 5000) {}

    Serial.println("Starting Keyboard test...");

    // Enable the USBHost 
    pinMode(PA_15, OUTPUT);
    digitalWrite(PA_15, HIGH);

    // if you are using a Max Carrier uncomment the following line
    // start_hub();

    while (!kbd.connect()) {
      Serial.println("No keyboard connected");        
        delay(5000);
    }

    kbd.attach(&kbd_key_cb);
    kbd.attach(&kbd_keycode_cb);
    kbd.attachRelease(&kbd_key_release_cb);
    Serial.println("End of Setup");
}

void loop()
{
    delay(1000);
}

void kbd_key_cb(uint8_t key) {
  Serial.print("Key pressed: ");
  Serial.print(key, HEX);
  Serial.print("(");
  if ((key >= ' ') && (key <= '~')) Serial.write(key);
  Serial.println(")");
}

void kbd_key_release_cb(uint8_t key) {
  Serial.print("Key released: ");
  Serial.print(key, HEX);
  Serial.print("(");
  if ((key >= ' ') && (key <= '~')) Serial.write(key);
  Serial.println(")");
}

void kbd_keycode_cb(uint8_t keycode, uint8_t mod) {
  Serial.print("Keycode: ");
  Serial.print(keycode, HEX);
  Serial.print(" mod: ");
  Serial.println(mod, HEX);
}
```
Now getting data:
```
End of Setup
Key pressed: 61(a)
Keycode: 4 mod: 0
Key released: 61(a)
Key pressed: 61(a)
Keycode: 4 mod: 0
Key released: 61(a)
Key pressed: 73(s)
Keycode: 16 mod: 0
Key released: 73(s)
Key pressed: 61(a)
Keycode: 4 mod: 0
Key pressed: 73(s)
Keycode: 16 mod: 0
Key pressed: 64(d)
Keycode: 7 mod: 0
Key released: 61(a)
Key released: 64(d)
Key released: 73(s)
Key pressed: 61(a)
Keycode: 4 mod: 0
Key pressed: 73(s)
Keycode: 16 mod: 0
Key pressed: 64(d)
```

Some interesting things to check: 
The RXHandler code was (maybe is) being called back a lot with 0 bytes transferred.  Not sure if that should be happening?

The USBHostKeyboard class is pretty limited and fragile.  I did hack up their handler code to reasonably handle a keyboard BOOT format setup.  First byte is modifiers, 2nd is ignored, and 3-8 are keycodes of keys currently pressed.   I also added in callback for when released.

But their conversion from keycod to code is really limited and can go BOOM.  That is it is a two dimension array
`static uint8_t keymap[4][0x39] = {`

With the first dimension is modifier, which is 8 bits:
<html>
<body>

Bit | Bit Length | Description
-- | -- | --
0 | 1 | Left Ctrl.
1 | 1 | Left Shift.
2 | 1 | Left Alt.
3 | 1 | Left GUI (Windows/Super key.)
4 | 1 | Right Ctrl.
5 | 1 | Right Shift.
6 | 1 | Right Alt.
7 | 1 | Right GUI (Windows/Super key.)

<br class="Apple-interchange-newline">
</body>
</html>
So use the Right shift will index into the 32nd index for the first dimension.
Or if you hit F12 that is keycode 0x45  which again is beyond the end of the table.

Also does not handle at all anything like Multimedia keys nor system keys.
Also does not handle setting and display and using keyboard settings like caps lock
...

Not sure how far to take it, may extract out into my secondary library, like the USBSerial adapter class and the like.
But at least something is working

`

AndrewCapon, 2023-11-12:
Nice one.

rxhandler being called with 0 bytes, is that every 1 ms?

AndrewCapon, 2023-11-12:
The keyboard is using USB interrupts:

```
if (dev && int_in) {
        host->interruptRead(dev, int_in, report, len_listen, false);
    }
```

So 1ms call to rxhandler.

mjs513, 2023-11-12:
Sorry for the late response but been a busy day today.  Just retested with Kurt's updated branch with my the xbox, switch pro, and ps3 and they are all still working.